brewpi-remix/brewpi-script-rmx #48: Brewer's Friend Integration

lbussy, 2019-07-09:
Integrate data into [Brewer's Friend](https://docs.brewersfriend.com/api/stream).

ianskate, 2019-09-04:
This would be great! I was just about to post a request for JSON output/custom endpoints with temperature probe logging to be sent to Brewfather.

Custom stream data/endpoints for that:
https://docs.brewfather.app/integrations/custom-stream
https://docs.brewfather.app/integrations/custom-api-endpoint

lbussy, 2019-09-04:
If I'm living right, the solution will work for most endpoints.  I'm working on a side project that needs some of this work so when I get to the end of that project, this one will start moving.

... my cat does the same thing. :)

lbussy, 2020-04-08:
An update for this:  I've been concentrating on some similar stuff (albeit in a different language) for Brew Bubbles.  What that's done is give me some ideas on how to best handle this, so when I get the next version of BB released, I will work on this.

Sorry it's taken so long.  "Soon" happens REALLY fast these days.