brewpi-remix/brewpi-script-rmx #65: updateFirmware.py fails due to errors in gitHubReleases.py

jsemroc, 2019-09-24:
updateFirmware.py fails during installation. Running it from the command-line  post-install produces the following error message:

Traceback (most recent call last):
  File "./updateFirmware.py", line 418, in <module>
    result = main()
  File "./updateFirmware.py", line 413, in main
    result = updateFromGitHub(beta, doShield, userInput)
  File "./updateFirmware.py", line 355, in updateFromGitHub
    localFileName = releases.getBin(tag, [board, shield, ".hex"])
  File "/home/brewpi/utils/gitHubReleases.py", line 146, in getBin
    fileName = self.download(downloadUrl, downloadDir)
  File "/home/brewpi/utils/gitHubReleases.py", line 73, in download
    uid = pwd.getpwnam(owner).pw_uid
NameError: global name 'stat' is not defined

Ultimately I discovered a couple of missing 'import' statements and a misnamed variable (learned a bit of python along the way too). 

Here's a patch file that can be applied with the command:

pi@raspberrypi:~ $ sudo patch ~brewpi/utils/gitHubReleases.py < ~/gitHubReleases.py.txt

[gitHubReleases.py.txt](https://github.com/brewpi-remix/brewpi-script-rmx/files/3649872/gitHubReleases.py.txt)

lbussy, 2019-09-24:
Thanks.  I’ll have a look when I get home tomorrow.  That’s a workflow which was well-tested.  I’m a little surprised you had a failure.

lbussy, 2019-09-26:
Well, not sure how that made it past my QC but it sure did.  I'll get that changed.  Thanks for the submission.

closed 2019-09-27 by lbussy