brewpi-remix/brewpi-script-rmx #95: Error flash Arduino Firmware

matko93, 2019-12-26:
I installed a clean version of Brewpi and when I try to flash the firmware on the Arduino I get this error:
sudo /home/brewpi/utils/updateFirmware.py

Stopping this chamber's instance(s) of BrewPi to check/update controller.

Stopping BrewPi with PID 848.

Checking current firmware version.
Opening serial port.
Traceback (most recent call last):
File "/home/brewpi/utils/updateFirmware.py", line 126, in updateFromGitHub
family = hwVersion.family
AttributeError: 'NoneType' object has no attribute 'family'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/home/brewpi/utils/updateFirmware.py", line 420, in
result = main()
File "/home/brewpi/utils/updateFirmware.py", line 415, in main
result = updateFromGitHub(beta, doShield, userInput)
File "/home/brewpi/utils/updateFirmware.py", line 135, in updateFromGitHub
"\nunresponsive, or if this is a new controller you can choose to proceed" +
File "/home/brewpi/utils/updateFirmware.py", line 56, in pipeInput
result = raw_input(prompt)
NameError: name 'raw_input' is not defined

lbussy, 2019-12-27:
> Same/similar issue. Automated brewpi install fails at arduino flash step (no error shown, the prompt went away too quickly). Tried manual install using updateFirmware.py command but it also failed: Error output below.

What you've posted there seems more like a transitive issue with either your Internet or GitHub.   I added a better error message in case this happens again.

lbussy, 2019-12-27:
> I installed a clean version of Brewpi and when I try to flash the firmware on the Arduino I get this error:

Testing a fix for this now.

lbussy, 2019-12-27:
Okay, I just merged a fix into master.  To update, execute the following:

``` bash
cd /home/brewpi
sudo /home/brewpi/utils/doUpdate.sh
```

Replace `/home/brewpi` with the root of your BrewPi install if you have placed it in a place other than the default.  If you are using multi-chamber, you must do this for each chamber's home.

When you've got that done, you can try to flash firmware again by issuing the following command:

```
sudo /home/brewpi/utils/updateFirmware.py
```

Please report back on your results.

matko93, 2019-12-28:
ok now i managed to flash arduino. But I can't connect it with brewpi. 

[image]

lbussy, 2019-12-28:
The script is not running.  Judging by the time you posted I'm guessing you are not in the US so I am going to give you a few things to do that should take care of multiple issues to cut down on the back and forth.  In the end, things should be running:

From the Raspberry Pi command line:

```
sudo /home/brewpi/utils/doPerms.sh
```

The output should look like:

```
 $ sudo /home/brewpi/utils/doPerms.sh

***Script doPerms.sh starting.***

Fixing file permissions for /var/www/html/chamber1.

Fixing file permissions for /home/brewpi/chamber1.

Allowing BrewPi python access to Bluetooth interfaces.

Checking user accounts.

***Script doPerms.sh complete.***
```

Then do:

```
sudo systemctl restart brewpi
```

There should be no output here.  If there is, let me know.  Then:

```
sudo reboot now
```

This will reboot your Pi.  When it reboots, refresh your web page and if the button in the top-right still says `Script not running!`, click the button (once) and it should start within 30 seconds.

This takes care of a lot of potential issues.  If at the end of all this the script will not start, I need to see the logs.  You can issue these commands and each will give you a link.  Run these and send me those links:

```
cat /home/brewpi/logs/stdout.txt | nc termbin.com 9999
cat /home/brewpi/logs/stderr.txt | nc termbin.com 9999
journalctl -u chamber1.service -b --no-pager | nc termbin.com 9999
```

You will get back a link from each that looks like this; share it here:

```
https://termbin.com/h38z
```

matko93, 2019-12-28:
This is the error he gave me:
https://termbin.com/5yg6
I tried to reinstall the script without configuring the Tilt and it is now working.
Thanks so much.

lbussy, 2019-12-28:
It looks like you don't have Bluetooth.  Are you running a Pi 2?  If so you need a Bluetooth dongle to use Tilt.

closed 2019-12-28 by lbussy