brewpi-remix/brewpi-script-rmx #101: Idle Status

matko93, 2019-12-29:
I can't get the script to work. The status remains in "Idle" even after 4 hours.
I have attached arduino relays and these work. 
I also try to set the "Assigned to" just don't want to save on "Beer 1". 
I 
[image]
[image]
[image]

lbussy, 2019-12-30:
It looks like you have your system set to “Beer Constant” but only have one temp sensor configured.   You need two for that.  If you only have one, use “Chamber Constant.”

Also, please see the documentation [here](https://brewpi-userdocs-rmx.readthedocs.io/en/latest/devices/index.html) for instructions on how to set up devices correctly.

lbussy, 2020-01-05:
I've not heard back from you on this.  If I don't hear back I'll assume you have it fixed and go ahead an close it.

closed 2020-01-16 by lbussy