brewpi-remix/brewpi-script-rmx #124: Unhanded Exception in brewpi.py causes script to hang

TChilderhose, 2020-05-30:
`brewpi.py` script just randomly hangs for me sometimes and doesn't recover. I was able to find a stack trace of the issue. Looks like you catch the one exception, but logging the exception causes it's own exception and gets stuck.

```
2020-05-30 15:50:47  Caught an unhandled exception.
Traceback (most recent call last):
  File "/home/brewpi/brewpi.py", line 716, in <module>
    conn.send('ack')
TypeError: a bytes-like object is required, not 'str'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/brewpi/brewpi.py", line 1506, in <module>
    e.errno, e.strerror, type, fname, traceback.tb_lineno
AttributeError: 'TypeError' object has no attribute 'errno'
```

I've replaced `e.errno` with `getattr(e, 'errno', '')` for now as it looks like `errno` is only valid for [OSError](https://docs.python.org/3/library/exceptions.html#OSError) Exceptions. I think that should probably be enough vs explicitly handling `OSErrors` in their own catch.

Also, I would love to help out with the project if you're looking for some. I'm a full-time Software Developer and have been itching to work on some sort of side/passion project in my free time.

lbussy, 2020-05-30:
That's ... sloppy of me.  I'll have a look in a few.

Sure I'd love the help.  I saw your PR earlier, I intend to get it in a dev branch and check it out when I'm done with some things on my TODO list on another project. 

For my not horribly organized project board, have a look [here](https://github.com/orgs/brewpi-remix/projects).  Just being one of me right now, not everything is perfectly laid out (not even marginally well-laid out for that matter) but at least I think all of them are in there.  I am NOT a software dev by trade, but I do work in the industry.  If you see something I can manage better, feel free to let me know that as well.

Given I've been a one-man band I have not done a brain dump for all of the improvements which could be made, but that's a future-Lee problem.

TChilderhose, 2020-06-01:
Awesome, sounds good. I'll check out the list and see if there is something I can help out with quickly, but having a few stability issues with my set up (Pi Zero W w/ Arduino Uno) so debugging that first. 

I think I've found one of the causes, I think my setup needs ` time_out=0` in `ser = util.setupSerial(config)` or I get a lot of exceptions trying to read from the controller. From the pySerial docs it says `timeout = 0: non-blocking mode, return immediately in any case, returning zero or more, up to the requested number of bytes` [link](https://pyserial.readthedocs.io/en/latest/pyserial_api.html). Looks like it was changed here https://github.com/brewpi-remix/brewpi-script-rmx/commit/369c94c2ac3132ee6a265ec836e1eefed2e05842. Are you using a Network controller? Just wondering if setting a Timeout is actually needed

Also, the `logError` function that is related to the `e.errno` has a formatting issue too. `logError("Error info:\n\tError: ({0}}): {1}}\n\tType: {2}\n\tFilename: {3}\n\tLineNo: {4}".format(` There are too many `}` closing the 0 and 1 format items. I can make fix PR for these Issue if you'd like.

lbussy, 2020-06-01:
In the words of another gentleman I collaborate with: "Never not make a PR."  :)

Most of the network-related changes were crowd-sourced from a thread on HBT.  I do not normally run one, so spurious issues go unnoticed until it has some real-world time.

Thanks for digging in.  Work was crazy last week and I'm running a sleep deficit.

TChilderhose, 2020-06-01:
Haha. Alright. I wont be able to work on it until tomorrow night, but I'll make a PR for it.

Ah okay. I'll go through the thread and read through the conversation. It's been awhile since I was on that thread.

No worries, glad to help. I had one of those weeks a few weeks ago so I get it haha.

lbussy, 2020-06-14:
Fixed and merged to devel.

closed 2020-06-14 by lbussy