brewpi-remix/brewpi-script-rmx #142: BlueBean Tilt Battery Readout

lbussy, 2020-12-05:
Think about how we can show a BlueBean-type battery indicator.

thorrak, 2020-12-05:
You're thinking only for the Gen 1 tilts, or are you thinking of doing something where you assume the number of weeks on a battery and use that to come up with a %?

lbussy, 2020-12-05:
Was exchanging email with a user who has a BlueBean and I pretty much wrote those off (after I sent yours back.)  I thought I might be able to add a conditional display between the new or old.

thorrak, 2020-12-05:
If you want to borrow it again, just let me know. (Preferably in the next week)

lbussy, 2020-12-05:
That would be cool - otherwise, I'm going to write stuff and ask you to test.  :)

lbussy, 2020-12-28:
This is still kicking my ass.  `gattool` has been removed now, and there's no way (yet) to use any of the current Python libs to get this information.

lbussy, 2021-04-19:
Not going to fix this.  Support has been sketchy and there's just not enough of these users.

closed 2021-04-19 by lbussy