brewpi-remix/brewpi-script-rmx #168: Refresh device list issue

wallem89, 2021-03-05:
I have an issue when refreshing the device list in the Device Configuration tab in the Maintenance Panel. If I refresh the script stops and it needs to restart. I had this issue before and it was bugging me from time to time. When I had to change a temperature sensor I had this issue every time I tried (even after updating BrewPi Remix to version 0.7.2. I took a snap from the stderr log:
```
2021-03-05 15:40:48 [E] Log files erased.
2021-03-05 15:49:42 [E] Caught an unexpected exception.
2021-03-05 15:49:42 [E] Error info:
2021-03-05 15:49:42 [E] Type:
2021-03-05 15:49:42 [E] Filename: brewpi.py
2021-03-05 15:49:42 [E] LineNo: 1117
2021-03-05 15:49:42 [E] Error: a bytes-like object is required, not 'str'
```

On this line [brewpi.py:1117](https://github.com/brewpi-remix/brewpi-script-rmx/blob/dcfe1306ba1235b4133d67bd2b15c499ecf532e0/brewpi.py#L1117) the following is done:
`phpConn.send("device-list-not-up-to-date")`.

I think this is incorrect since the send function expects a byte-like object instead of a string. I have changed the code to:
`phpConn.send(str.encode("device-list-not-up-to-date"))`

Now I can refresh the device list every time without crashing the script!

It is a bit strange though because I never see the text "device-list-not-up-to-date".

lbussy, 2021-03-05:
That tracks with something I noted a while back but I am not sure if I pushed the fix.  I'll check as soon as my client quits expecting me to work. 😄

lbussy, 2021-03-10:
Okay, I had a look at this finally, and I cannot reproduce the issue.  What version of the firmware are you using?

Is it possible that the error was just there in the log from a previous issue?

The only time you should hit that is if you get an error from the device list to begin with.

lbussy, 2021-03-13:
> I can off course change that line of code again to reproduce the issue but what information can I provide to be able to look in to it?

I'd be interested to see the complete `stderr.txt` and `stdout.txt` (extra points for clearing them beforehand of course).  Snagging the daemon log may be helpful as well:

```
pi@brewpi:~ $ journalctl -u brewpi -b
-- Logs begin at Thu 2021-03-11 13:17:01 CST, end at Sat 2021-03-13 13:00:19 CST
Mar 11 14:01:42 brewpi systemd[1]: Started BrewPi Remix daemon for: brewpi.
Mar 11 14:01:50 brewpi brewpi[597]: Logging to /home/brewpi/logs/.
Mar 11 14:01:50 brewpi brewpi[597]: Output will not be shown in console.
Mar 11 14:01:50 brewpi brewpi[597]: Starting BrewPi.
```

lbussy, 2021-03-24:
Aha!  I did catch it.  Not sure why I missed it but that's not important right now. :P

wallem89, 2021-03-24:
> Aha! I did catch it. Not sure why I missed it but that's not important right now. :P

Okay good to hear.

I didn't had time to extract the additional data you asked for. And then I had a brew day so now my climate chamber is busy keeping the wort warm enough. So I didn't want to mess around it now.

lbussy, 2021-03-24:
I get it - I actually found it starting a temp control session so it's ironic justice I guess. :)

lbussy, 2021-03-25:
Released to 0.7.3.

closed 2021-03-25 by lbussy