brewpi-remix/brewpi-script-rmx #174: Smooth or expire iSpindelSG values

pattew, 2021-04-13:


pattew, 2021-04-15:
I think I found out why i get the null values. I have my iSpindel to send message every 30 minutes and the variabel to clear old iSpindel values are timeoutiSpindel = 1800. So I guess to fix my problem I can either increase that variabel or send message more frequent from my iSpindel...

lbussy, 2021-04-15:
Hey, that's a good catch.  I can see why using a longer timeout is desirable for iSpindel.  I can try a change in the devel branch if you're willing to do a little fiddling to change branches?

pattew, 2021-04-16:
Thank you for increasing the timeout. I was thinking to improve it further, one could update the timeout value with the interval value the iSpindel is sending for example 3*interval 
Just an idea...

lbussy, 2021-04-16:
I don't think the interval is part of what we receive, is it?  The timeout is sort of protection against not receiving a packet, so depending on the timing of the packet to calculate the interval is sort of self-defeating. :)

I could make it configurable, easily enough.

pattew, 2021-04-16:
Yes actually it is, here is a copy of the json
`{"name": "iSpindel001", "ID": 12773660, "angle": 29.76373, "temperature": 20.8125, "temp_units": "C", "battery": 5.338894, "gravity": 1.008273, "interval": 1800, "RSSI": -58}`
But it was just an idea, I have the tendenssie to think out loud.... So for me it is good enough as it is.

lbussy, 2021-04-16:
No that's great, I can do that easy enough.  I guess I could have just looked at the damned thing. :)

lbussy, 2021-04-17:
Okay, I'm setting the timeout to 3.5* the interval.

Do you know how to swap branches or would you like some instructions?

pattew, 2021-04-17:
Thank you! I actually work with git on daily bases so I think i will manage. 
And thank you for the great job you are doing with this project.

lbussy, 2021-04-19:
Added:

- 962ffd3 Add log message when re-setting timeout

lbussy, 2021-04-19:
Going to merge these to devel.

pattew, 2021-04-19:
When I changed branch to devel I seem to loose contact with the arduino. Do you have any idea why?

lbussy, 2021-04-19:
> When I changed branch to devel I seem to loose contact with the arduino. Do you have any idea why?

I suspect you need to run the permissions script:

`/home/brewpi/utils/doPerms.sh`

pattew, 2021-04-19:
> 
> 
> > When I changed branch to devel I seem to loose contact with the arduino. Do you have any idea why?
> 
> I suspect you need to run the permissions script:
> 
> `/home/brewpi/utils/doPerms.sh`

I have done that. But I have not had time to fiddle with i much. I have to test a little bit more later.

lbussy, 2021-04-19:
It's generally that.  Sometimes the daemon will stop after a number of failures.  You can check ith with:

`sudo systemctl status brewpi`

If it's stopped, restart it with:

`sudo systemctl restart brewpi`

pattew, 2021-04-20:
It was a problem with the new python virtual enviroment, changed in the last version. I had to install packaging into the venv folder also. Now I have changed the script branch to ispindel_timeout and testing hopefully no null reading any more :)

lbussy, 2021-04-20:
I need to add packaging to the venv?  Or do you mean I did add it and your venv did not have it?

You can always run the utility scripts if you start having weird issues.  The [README in that folder](https://github.com/brewpi-remix/brewpi-script-rmx/tree/master/utils) explains what does what.  In the case you had, doDepends (for dependencies) would have cleaned that up, followed by doPerms of course.

pattew, 2021-04-20:
> I need to add packaging to the venv? Or do you mean I did add it and your venv did not have it?
>
> You can always run the utility scripts if you start having weird issues. The [README in that [...]

I tried running the scripts but for some reason the packaging was missing, but I probably did something wrong. I just wanted to let you know what my problem was if someone else has the same problem.

lbussy, 2021-04-20:
That makes sense.  I added `packaging==20.9` to `requirements.txt`, but since you didn't actually "upgrade" (which would have run `doDepends.sh` and cleaned it up) you did not get that package.

pattew, 2021-04-22:
Everthing was looking good but then we hadd a short poweroutage and I got a null reading again and the graph crashes. Then I looked a little bit on the json and noticed a comma after the null value. In the brewpiJson.py file line 129 there is a comma after the null value that makes the parsJson function throw an exception. Removing that comma will fix it. Hopefully :)
`
elif iSpindel:
        if row['spinSG'] is None:
            jsonFile.write(",null,") 
        else:
           jsonFile.write(",{\"v\":" + str(row['spinSG']) + "}")
`

lbussy, 2021-04-22:
Can you share the portion of the log that caused the issue, please?

No, I didn't test power outages. :)

pattew, 2021-04-22:
Here is two rows first with null value second with ok value. If you look at the first row at the end after the null value there is a comma that should not be there it makes it a invalid json file. This is actually the bug that I was after in the first place. That the graph crashes if iSpindel has null value. I thought it was due to the null but it was due to the extra comma in the row if it is null.  
`{"c":[{"v":"Date(2021,3,20,13,8,2)"},{"v":18.19},null,null,{"v":18.5},{"v":5.0},null,{"v":19.25},{"v":4},null,]},
{"c":[{"v":"Date(2021,3,20,13,10,29)"},{"v":18.22},null,null,{"v":18.5},{"v":18.0},null,{"v":19.25},{"v":4},{"v":1.175}]},`
I have removed the comma in my code and now it works even if it is null.
So if you change row 129 from this `jsonFile.write(",null,")` to this `jsonFile.write(",null")` it will also work after a outage  :)
Sorry for being so difficult :)

lbussy, 2021-04-22:
I love the idea of an open-source solution.  I support Tilt with my $$ though.  I do some things because I enjoy fiddling, but there are some things I'll trust to someone else (and can therefore complain if it does not work.)

I think I'm well over 12 months on a couple of my Tilts now, the battery life is fantastic.

pattew, 2021-04-22:
If you want to test it you can connect a iSpindel and start a log and then restart the brewpi server then the value is always null and the graph will crash. The iSpindel intervall need to bee at least 5 min so it doesn't update the value to soon.

lbussy, 2021-04-22:
Check the `more_ispindel` branch and see if that gets it for you.

I need to create an iSpindel simulator.  A gent created one for the Tilts and it was a life-changer for testing.

pattew, 2021-04-22:
I love to build things and especially if it gives me better tasting beer :) . I am also building a counter pressure beer bottle filler where I use a arduino mega. It purges the bottle and start filling and stops when it's full. I have some improvement still but I have probably filled a couple of hundreds bottles with it so far. a little of topic :)
[image]

thorrak, 2021-04-22:
Oh my god. My jaw just dropped.

That's amazing!

pattew, 2021-04-22:
Thank you :) 
I don't have so much extra time for the moment but if I have some spare I will have a look at the more_ispindel rep

lbussy, 2021-04-22:
I've merged this into `devel` so you can try that.

pattew, 2021-04-23:
> 
> 
> I've merged this into `devel` so you can try that.

I pulled down the changes and it works like charm even after a reboot. Great thank you!

lbussy, 2021-04-24:
Awesome.  I have some more changes going into devel so I'm likely to release to master again here soon.  Going to close this one now since it's in devel and fixed.  I need a win today. :)

PR #175 (2021-04-19): #174 increase timeout
Potentially closes #174.
closed 2021-04-24 by lbussy