brewpi-remix/brewpi-script-rmx #179: nginx Detection and Handling with Docker

lbussy, 2021-04-19:
When nginx is run in Docker, the script will still see it's installed, but it will fail to reconfigure it for an alternative port.  I need to allow installing Apache2 to an alternative port in these cases.

lbussy, 2021-04-19:
Current flow fails in this manner:

```
Checking for previously installed nginx packages.

Found nginx packages installed. nginx will interfere with Apache2 and it is
recommended to uninstall. You can either do that now, or choose to reconfigure
nginx to use an alternate port. Choose one of the following:

        [u] Uninstall nginx (this will break Fermentrack if installed.)
        [r] Reconfigure nginx (and Fermentrack) to use a different port.
        [X] Exit and do nothing.

[u/r/X]: r

Reconfiguring nginx to alternate port.

Attempting to configure nginx for ports 81/444.
cp: cannot stat '': No such file or directory
sed: can't read : No such file or directory
sed: can't read : No such file or directory
sed: can't read : No such file or directory
sed: can't read : No such file or directory
Failed to restart nginx.service: Unit nginx.service not found.

Reconfigured nginx to serve applications on port 81/444. You will have to
access your previous nginx websites with the port at the end of the URL like:
http://mule.local:81 or http://172.17.0.1
192.168.1.11:81
```

Need to:

- [ ] Add choice to allow configuring Apache to non-standard port
- [ ] Add choice to ignore dire warnings and charge ahead to likely failure
- [ ] Fail (or do something helpful) if nginx cannot be configured
- [ ] Possibly be even more helpful if we find the port in use but no service present (e.g. Docker)

thorrak, 2021-04-19:
How are you detecting nginx?

lbussy, 2021-04-19:
With: `$(sudo netstat -tulpn | grep :80 | grep nginx)`

PR #181 (2021-04-20): Nginx detect
Closes #179.

Contains:

- 6d0f336 Fix nginx detection
- 7ee759f Cleanup variables
closed 2021-04-22 by lbussy