brewpi-remix/brewpi-script-rmx #200: Installation fails with venv "--prompt bpr" error

TheFiZi, 2022-07-10:
Did some quick greps for `--prompt bpr` in the installation files to see if I could make a quick hack to fix the script, no luck

Thanks for keeping this alive!

```
me@nutmeg:~ $ curl -L raw.githubusercontent.com/brewpi-remix/brewpi-tools-rmx/master/bootstrap.sh | sudo bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 18822  100 18822    0     0   117k      0 --:--:-- --:--:-- --:--:--  117k

***Script bootstrap.sh starting.***

Running on: Raspbian GNU/Linux 9 (stretch)

               ___                ___ _   ___           _
              | _ )_ _ _____ __ _| _ (_) | _ \___ _ __ (_)_ __
              | _ \ '_/ -_) V  V /  _/ | |   / -_) '  \| \ \ /
              |___/_| \___|\_/\_/|_| |_| |_|_\___|_|_|_|_/_\_\
════════════════════════════════════════════════════════════════════════════════
You will be presented with some choices during the install. Most frequently
you will see a 'yes or no' choice, with the default choice capitalized like
so: [y/N]. Default means if you hit <enter> without typing anything, you will
make the capitalized choice, i.e. hitting <enter> when you see [Y/n] will
default to 'yes.'

Yes/no choices are not case sensitive. However; passwords, system names and
install paths are. Be aware of this. There is generally no difference between
'y', 'yes', 'YES', 'Yes'; you get the idea. In some areas you are asked for a
path; the default/recommended choice is in braces like: [/home/brewpi].
Pressing <enter> without typing anything will take the default/recommended
choice.

Press any key when you are ready to proceed.

The time is currently set to Sat  9 Jul 15:48:50 PDT 2022.
Is this correct? [Y/n]:

Updating any expired apt keys.

Fixing any broken installations.
Reading package lists...
Building dependency tree...
Reading state information...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Checking and installing required dependencies via apt.

Cloning brewpi-tools-rmx repo.
Cloning into '/home/me/brewpi-tools-rmx'...
Already on 'main'
Your branch is up-to-date with 'origin/main'.

***Script install.sh starting.***

Checking for connection to GitHub.

Connection to GitHub ok.

Disk usage is 5%, free disk space is 27436 MB.

If you would like to use BrewPi in multi-chamber mode, or simply not use the
defaults for scripts and web pages, you may choose a name for sub directory and
devices now.  Any character entered that is not [a-z], [0-9], - or _ will be
converted to an underscore.  Alpha characters will be converted to lowercase.
Do not enter a full path, enter the name to be appended to the standard path.

Enter device/directory name or hit enter to accept the defaults.
[<Enter> = Single chamber only]:

Using '/home/brewpi' for scripts directory.

Scripts will use default 'port = auto' setting.

Creating and configuring accounts.

Cloning BrewPi scripts to /home/brewpi.
Cloning into '/home/brewpi'...
Already on 'main'
Your branch is up-to-date with 'origin/main'.

***Script doDepends.sh starting.***

Fixing any broken installations before proceeding.
Reading package lists...
Building dependency tree...
Reading state information...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Checking for previously installed php5 packages.

No php5 packages found.

Checking for previously installed nginx packages.

No nginx daemon found running on port 80.

Fixing any broken installations before proceeding.
Reading package lists...
Building dependency tree...
Reading state information...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Checking and installing required packages via apt.

No apt packages were missing.

No apt updates to apply.

Downloading current boards configuration.

Modifying UART speeds for BLEacon support.

Setting up venv for BrewPi user.
usage: venv [-h] [--system-site-packages] [--symlinks | --copies] [--clear]
            [--upgrade] [--without-pip]
            ENV_DIR [ENV_DIR ...]
venv: error: unrecognized arguments: --prompt bpr

*** ERROR ERROR ERROR ERROR ERROR ***
-------------------------------------
See above lines for error message.
Setup NOT completed.

*** ERROR ERROR ERROR ERROR ERROR ***
-------------------------------------

See above lines for error message.
Setup NOT completed.
```

lbussy, 2022-07-10:
Thanks for reporting this.

As an aside, a "fresh installation of _Stretch_" is a bit of an oxymoron.  Technically it is tagged "oldoldstable" by the maintainers and is in LTS support.  That said, it's version 9, and the current version is version 11, "_Bullseye_."  The current installer was written under _Buster_ I believe.  If for no other reason than to get current with all applicable security patches (including supporting libs, not just the OS,) I'd recommend flashing a fresh(er) image.

Neither shellcheck nor adhoc testing show any negative impact from this change:

``` bash
(bpr) brewpi@brewpi:~ $ USERROOT="/home/brewpi"
(bpr) brewpi@brewpi:~ $ venvcmd="python3 -m venv "$USERROOT/venv" --prompt bpr"
(bpr) brewpi@brewpi:~ $ echo $venvcmd
python3 -m venv /home/brewpi/venv --prompt bpr
(bpr) brewpi@brewpi:~ $ venvcmd="python3 -m venv $USERROOT/venv --prompt bpr"
(bpr) brewpi@brewpi:~ $ echo $venvcmd
python3 -m venv /home/brewpi/venv --prompt bpr
(bpr) brewpi@brewpi:~ $ 
```

lbussy, 2022-07-10:
Fixed in `devel`.

TheFiZi, 2022-07-10:
How do I pull the devel branch with the installer scripts so I can test it for you?

I'm also happy to just rebuild using buster or bullseye. When I read the docs before starting, this page (https://docs.brewpiremix.com/en/latest/rpi/index.html?highlight=stretch#basic-configuration-of-your-raspberry-pi) lead me to believe I needed to use stretch.

> This step by step guide will help you to set everything up on your Raspberry Pi and Arduino. Current technology is assumed throughout. **At the time of writing, this is a Raspberry Pi 3B+ (although just about any should work) running Raspbian Stretch.** If you deviate from that, you should know enough to help translate these instructions on your own use.

I never found any references explicitly telling me to use anything else.

lbussy, 2022-07-10:
Use:

``` bash
curl -L devinstall.brewpiremix.com | sudo bash
```

I guess I need to update that, huh?  :)

I test with whatever is current.

TheFiZi, 2022-07-10:
Same error

```
Setting up venv for BrewPi user.
usage: venv [-h] [--system-site-packages] [--symlinks | --copies] [--clear]
            [--upgrade] [--without-pip]
            ENV_DIR [ENV_DIR ...]
venv: error: unrecognized arguments: --prompt bpr

*** ERROR ERROR ERROR ERROR ERROR ***
-------------------------------------
See above lines for error message.
Setup NOT completed.

*** ERROR ERROR ERROR ERROR ERROR ***
-------------------------------------

See above lines for error message.
Setup NOT completed.
```

I wouldn't worry about it. I can just wipe/reload to bullseye. I'm going to guess that venv is too old and doesn't actually support `--prompt`

Checking https://docs.python.org/3/library/venv.html I see this help text:

```
usage: venv [-h] [--system-site-packages] [--symlinks | --copies] [--clear]
            [--upgrade] [--without-pip] [--prompt PROMPT] [--upgrade-deps]
            ENV_DIR [ENV_DIR ...]
```

The one being output with the error is missing `--prompt`

```
usage: venv [-h] [--system-site-packages] [--symlinks | --copies] [--clear]
            [--upgrade] [--without-pip]
            ENV_DIR [ENV_DIR ...]
```

TheFiZi, 2022-07-11:
I am up and running with Bullseye on my Raspberry Pi 3 Model B Rev 1.2

Thanks!

closed 2022-07-11 by TheFiZi

lbussy, 2022-07-11:
I guess it's gotta be an issue with the older version of `venv` in Stretch.  All is well that ends well, however.  Glad you got it running.