brewpi-remix/brewpi-script-rmx #202: Unable to install devices

dlrdlrdlr, 2023-02-25:
I'm using brewpiremix on an arduino uno installed with [this ](https://github.com/brewpi-remix/brewpi-tools-rmx/issues/65) fix due to git problems and have been unable to add a device. I have 2 temp probes on pin a4 and pins 5 and 6 going into a relay for a total of 4 devices. Oddly 6 are detected by the interface. The 4 expected ones as well as a digital pin on 2 and a5. When I attempt to set the slot and function of any of these the settings I selected disappear after I hit apply.  I have restarted the Pi, uninstalled and reinstalled BrewPi but have been unable to get devices to correctly show up. The Temps seem accurate so there is some communication there but I can't install them.

[image]

`Device list updated for uno with a revC shield.
Device config command sent, U:{"i":3,"c":1,"b":0,"f":2,"h":1,"p":2,"x":1}
No installed devices found.
Parsing available devices: Parsing device: {"i":-1,"t":0,"c":1,"b":0,"f":0,"h":2,"d":0,"p":18,"v":67.211,"a":"284A8481E3193C46","j":0,"nr":0}
Parsing device: {"i":-1,"t":0,"c":1,"b":0,"f":0,"h":2,"d":0,"p":18,"v":66.424,"a":"28492681E3473C04","j":0,"nr":1}
Parsing device: {"i":-1,"t":0,"c":1,"b":0,"f":0,"h":1,"d":0,"p":2,"x":1,"nr":2}
Parsing device: {"i":-1,"t":0,"c":1,"b":0,"f":0,"h":1,"d":0,"p":5,"x":1,"nr":3}
Parsing device: {"i":-1,"t":0,"c":1,"b":0,"f":0,"h":1,"d":0,"p":6,"x":1,"nr":4}
Parsing device: {"i":-1,"t":0,"c":1,"b":0,"f":0,"h":1,"d":0,"p":19,"x":1,"nr":5}
Device list updated for uno with a revC shield.
Device config command sent, U:{"i":4,"c":1,"b":0,"f":2,"h":1,"p":19,"x":1}
No installed devices found.
Parsing available devices: Parsing device: {"i":-1,"t":0,"c":1,"b":0,"f":0,"h":2,"d":0,"p":18,"v":66.986,"a":"284A8481E3193C46","j":0,"nr":0}
Parsing device: {"i":-1,"t":0,"c":1,"b":0,"f":0,"h":2,"d":0,"p":18,"v":66.313,"a":"28492681E3473C04","j":0,"nr":1}
Parsing device: {"i":-1,"t":0,"c":1,"b":0,"f":0,"h":1,"d":0,"p":2,"x":1,"nr":2}
Parsing device: {"i":-1,"t":0,"c":1,"b":0,"f":0,"h":1,"d":0,"p":5,"x":1,"nr":3}
Parsing device: {"i":-1,"t":0,"c":1,"b":0,"f":0,"h":1,"d":0,"p":6,"x":1,"nr":4}
Parsing device: {"i":-1,"t":0,"c":1,"b":0,"f":0,"h":1,"d":0,"p":19,"x":1,"nr":5}
Device list updated for uno with a revC shield.`

thorrak, 2023-02-25:
Have you tried resetting the EEPROM of the Uno?

dlrdlrdlr, 2023-02-25:
That seems to have fixed it thank you.

closed 2023-02-25 by dlrdlrdlr

lbussy, 2023-02-25:
The three most common fixes are reset permissions, get a better power supply, and erase EEPROM.  Glad you got it going,.