brewpi-remix/brewpi-script-rmx #208: BrewPi remix won't load/connect script

shipcoma, 2023-11-20:
Just did a fresh install of BrewPi remix on a new Pi 4 with the Arduino I already had. Disconnected all the relays/temp probes from the Arduino before install and allowed the BrewPi remix installer to flash the Arduino, then reconnected and assigned all the devices in the maintenance panel. During this phase it looked like the script was running OK (as you can't access devices unless the script is running I think).

I can now access the BrewPi remix page and temperatures are being monitored fine. But even though the script looked like it was connecting OK during device setup there is now a problem. Trying to set either beer constant mode or a temp profile, the status bar at the top of the is stuck at "script is starting up". In the logs I am seeing repeated failed attempts to run the script like this:

> 2023-11-20 15:32:42 [E] Starting BrewPi.
> /home/brewpi/brewpi.py:751: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
> if LooseVersion(hwVersion.toString()) < LooseVersion(compatibleHwVersion):
> 2023-11-20 15:33:59 [E] Starting BrewPi.
> /home/brewpi/brewpi.py:751: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
> if LooseVersion(hwVersion.toString()) < LooseVersion(compatibleHwVersion):
> 2023-11-20 15:34:10 [E] Caught an unexpected exception.
> 2023-11-20 15:34:10 [E] Error info:
> 2023-11-20 15:34:10 [E] Type:
> 2023-11-20 15:34:10 [E] Filename: brewpi.py
> 2023-11-20 15:34:10 [E] LineNo: 1779
> 2023-11-20 15:34:10 [E] 

> Error: invalid mode: 'rU'

And if I now go back into the device setup page and refresh I see:

> Error while receiving device configuration: SyntaxError: Unexpected token 'C', "Couldn'tre"... is not valid JSON

Any ideas what's causing this please?

lbussy, 2023-11-20:
Would you please issue this command:

```bash
sed -n 1779p /home/brewpi/brewpi.py
```

... and let me know what it returns?

shipcoma, 2023-11-20:
newTemp = temperatureProfile.getNewTemp(util.scriptPath())

lbussy, 2023-11-20:
Are you on HBT where you can send me the whole log?

shipcoma, 2023-11-20:
Just joined - shall I dump a load of output from the stderr and stout in a new post in the mega-thread?
Thanks

shipcoma, 2023-11-20:
Ah, can't PM yet as not enough posts! Will put in the forum?

lbussy, 2023-11-20:
WHat's your user name there?

shipcoma, 2023-11-20:
shipcoma

lbussy, 2023-11-20:
Gotcha, you should be able to reply there

lbussy, 2023-11-21:
Closing as a data issue.

For anyone browsing old issues:  There seems to have been an issue with the .CSV files holding the profile.

closed 2023-11-21 by lbussy