brewpi-remix/brewpi-script-rmx #210: Issue with the .CSV files holding the profile redux

mikeohio, 2023-12-15:
On issue #208 your resolution is the following: 
           Closing as a data issue.

For anyone browsing old issues:  There seems to have been an issue with the .CSV files holding the profile.

_Originally posted by @lbussy in https://github.com/brewpi-remix/brewpi-script-rmx/issues/208#issuecomment-1820833128_

I'm having the exact same problem.  Would you be able to guide me how to fix the .CSV data file issue?  

            
[image]

lbussy, 2023-12-15:
The profile CSV file is not proper in some way. Granted, the script should probably do a better job with a malformed profile, but this portion is pretty old, so at least it's not my fault.  :)

Here are the steps I recommended to him to get running:

* `sudo systemctl stop brewpi`
* `sudo apt-get install screen`
* `sudo screen /dev/ttyACM0 57600`
* type `E`
* Exit screen by typing ctrl a, k
* `sudo systemctl stop brewpi`

This will wipe your controller's configuration (devices), so be sure you are ready.  Delete or fix the CSV files before switching it back to a profile.

mikeohio, 2023-12-19:
Thank you so very much.  I have it running, it is monitoring and logging temps and it is controlling my heater and cooler. 
I read through a number of your previous posts and found that you favored the raspberry pi's 'lite' installs. Just in case anyone else experiences a similar issue, when I used the 'full' raspberry pi installations, my brewpi installation returned the above error. The 'lite' versions works great.  
[image]

bdcapps, 2024-05-07:
Having this same issue. The problem was not the CSV files. In a brand new install with Python 3.11.2, using the 'rU' flag is causing this issue on line 41 of temperatureProfile.py. Changing this to  

`with open(util.addSlash(scriptPath) + 'settings/tempProfile.csv', 'r') as csvfile:`

fixed the issue for me.

lbussy, 2024-05-16:
Merged direct to main.  Living large!  :)

closed 2024-05-16 by lbussy