brewpi-remix/brewpi-script-rmx #214: Issue with DS18B20 Sensor Detection and Persistence in BrewPi Remix

ultrasaha, 2025-03-16:
Problem Description:
I am using BrewPi Remix on original 2014 hardware (details below). Initially, the DS18B20 temperature sensors were detected, but one of them was not properly saved. Every time BrewPi restarted, I had to manually go into settings and press "Apply" for the second sensor to work again.

To try to fix this, I performed an "Advanced Settings -> Setting Arduino to Factory Defaults". Now, in "Detected Devices," all three sensors appear, but when assigning them to a slot and pressing "Apply," only one is saved in "Installed Devices." The other two appear as "ghost sensors" with no temperature readings.

I have no advanced knowledge of Raspberry Pi or programming, so I would appreciate guidance on whether there is a way to fix this issue or if upgrading my hardware would be the best solution.

Steps to Reproduce:

Install BrewPi Remix on a Raspberry Pi Model B+ with Raspberry Pi OS Lite.
Connect DS18B20 sensors and detect them in BrewPi.
Assign them to slots and press "Apply."
Only one sensor successfully saves in "Installed Devices," while the others appear but do not register a temperature.
Hardware Used:

Sainsmart Uno R3
Raspberry Pi Model B+ (2014)
DS18B20 OneWire digital temperature sensors
Bluetooth dongle for Tilt
BrewPi Remix with Tilt and Tilt Pi support
Software Installed:

Raspberry Pi OS Lite (latest version)
BrewPi Remix
Tilt and Tilt Pi
Previous Attempts to Fix:

Connected a third sensor to rule out a hardware failure.
Performed a factory reset on the Arduino settings.
Reinstalled BrewPi Remix.
Main Question:
Is there a way to fix this issue without upgrading the hardware? Could it be due to incompatibility with older hardware?

Thanks in advance for any help!

[image]

lbussy, 2025-03-16:
It sounds like the Arduino is faulty - have you tried another?

ultrasaha, 2025-03-18:
Just installed the new Arduino and reinstalled brewpi remix, but the problem persists:

[image]

Any ideas?

lbussy, 2025-03-18:
Generally the "erase controller" (or whatever it's called, I am on my phone) fixes that.   Understand it's a new Arduino, but try that again. 

If that does not help, seeing your error and normal log from /home/brewpi/logs from memory, is what will get us going the right direction.

lbussy, 2025-03-19:
Okay, I am on my laptop now.

Do the "Advanced Settings -> Setting Arduino to Factory Defaults."   This erases the EEPROM area of the Arduino, which is not affected by re-flashing the firmware.  This is sort of a known issue with Arduinos in general.

You can also try the Arduino test firmware, which will validate it sees all of your probes and then erase the EEPROM.  You will have to re-flash the BrewPi firmware again after this test.  Instructions for the test are here:

https://github.com/brewpi-remix/uno-test

If you do not have an LCD attached, just let it sit for about five minutes.  It takes nowhere near that time but just to be sure.  Or as a sort of advanced extra credit thing, after you flash the firmware, issue this command:

`cat /dev/ttyUSB0` (replace the port with the one you use above.). Exit with Ctrl-C

I've never tried the cat thing above but it looks like it will work.  If not, install screen:

`sudo apt-install screen`

Connect to the comm port with:

`screen /dev/ttyUSB0 57600`

To exit screen, press Ctrl+A, then K, and confirm with Y.  Again, make sure the port you use is correct.

With either method, you can hit the reset button on the Arduino, and the debug will restart and print will appear on your screen.

Finally, the logs in `/home/brewpi/logs/stderr.txt` and `/home/brewpi/stdout.txt` (after you try to assign sensors again) will show us if there's a script error.

ultrasaha, 2025-03-21:
Hi, sorry for the delay. It's been 2 days to reset everything but still not working. I connected the Arduino to the computer using Arduino IDE to try to test the temperature probes, but I fail and I didn't know how to restablish everything. But I managed somehow...
Now I did "Advanced Settings -> Setting Arduino to Factory Defaults."
From what I see one temperature probe (chamber) is working fine. But I can not manage that recognizes the beer probe, when I try to apply It creates the "ghost probe".
(stderr.txt)

pi@ultrasaha:~ $ cat /home/brewpi/logs/stderr.txt
2025-03-20 20:15:17 [E] Starting BrewPi.
/home/brewpi/Tilt.py:328: RuntimeWarning: coroutine 'BLEScanRequester.send_scan_request' was never awaited
  self.btctrl.send_scan_request()
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
/home/brewpi/brewpi.py:751: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(hwVersion.toString()) < LooseVersion(compatibleHwVersion):
2025-03-20 20:15:28 [E] Failed to retrieve Blue Tilt value, restarting Tilt.
/home/brewpi/Tilt.py:343: RuntimeWarning: coroutine 'BLEScanRequester.stop_scan_request' was never awaited
  self.btctrl.stop_scan_request()
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
/home/brewpi/Tilt.py:345: RuntimeWarning: coroutine 'BLEScanRequester.send_command' was never awaited
  self.btctrl.send_command(command)
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
2025-03-20 20:17:37 DOWIFI ERROR: Unable to determine wireless interface name.  Exiting.
2025-03-20 20:17:41 DOWIFI ERROR: Unable to determine wireless interface name.  Exiting.
2025-03-20 20:17:45 DOWIFI ERROR: Unable to determine wireless interface name.  Exiting.
2025-03-20 20:17:49 DOWIFI ERROR: Unable to determine wireless interface name.  Exiting.
2025-03-20 20:17:53 DOWIFI ERROR: Unable to determine wireless interface name.  Exiting.
2025-03-20 20:17:56 DOWIFI ERROR: Unable to determine wireless interface name.  Exiting.
2025-03-21 16:14:36 [E] Starting BrewPi.
/home/brewpi/Tilt.py:328: RuntimeWarning: coroutine 'BLEScanRequester.send_scan_request' was never awaited
  self.btctrl.send_scan_request()
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
/home/brewpi/brewpi.py:751: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(hwVersion.toString()) < LooseVersion(compatibleHwVersion):
2025-03-21 16:14:47 [E] Failed to retrieve Blue Tilt value, restarting Tilt.
/home/brewpi/Tilt.py:343: RuntimeWarning: coroutine 'BLEScanRequester.stop_scan_request' was never awaited
  self.btctrl.stop_scan_request()
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
/home/brewpi/Tilt.py:345: RuntimeWarning: coroutine 'BLEScanRequester.send_command' was never awaited
  self.btctrl.send_command(command)
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
2025-03-21 16:16:48 [E] Failed to retrieve Blue Tilt value, restarting Tilt.
2025-03-21 16:18:49 [E] Failed to retrieve Blue Tilt value, restarting Tilt.
2025-03-21 16:20:50 [E] Failed to retrieve Blue Tilt value, restarting Tilt.
2025-03-21 16:22:51 [E] Failed to retrieve Blue Tilt value, restarting Tilt.

(stdout.txt)

2025-03-21 16:36:03 [N] Cannot process line from controller:
pi@ultrasaha:~ $ PuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTY

At this point I'm not sure if the physical probe is failing.

I also bought a new version of Raspberry. Should I give up on the old one?

Many thanks.

ultrasaha, 2025-03-21:
I reinstalled brewpi without adding a TILT.
Done "Advanced Settings -> Setting Arduino to Factory Defaults."
Temperature probe (chamber) is working . Beer probe not installing, when  trying to apply It creates the "ghost probe".

Logs wihtout TILT:

pi@ultrasaha:~ $ cat /home/brewpi/logs/stderr.txt
2025-03-21 18:50:22 [E] Starting BrewPi.
/home/brewpi/brewpi.py:751: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(hwVersion.toString()) < LooseVersion(compatibleHwVersion):

pi@ultrasaha:~ $ cat /home/brewpi/logs/stdout.txt
2025-03-21 18:50:23 [N] 0.8.1 (main) [a805939]
2025-03-21 18:50:23 [N] Starting 'My BrewPi Remix Run.'
2025-03-21 18:50:23 [N] Opening serial port.
2025-03-21 18:50:23 [N] Waiting 10 seconds for board to restart.
2025-03-21 18:50:33 [N] Checking software version on controller.
2025-03-21 18:50:33 [N] Found BrewPi v0.2.10 build unknown, running on an Arduino Uno with a revC shield on port /dev/ttyACM0.
2025-03-21 18:50:33 [N] Warning: version number of local copy of logMessages.h
2025-03-21 18:50:33 [N] does not match log version number received from
2025-03-21 18:50:33 [N] controller. Controller version = 1, local copy
2025-03-21 18:50:33 [N] version = 3.
2025-03-21 18:50:33 [N] New JSON received from controller.
2025-03-21 18:50:39 [N] Installed devices received: [{"i": 0, "t": 1, "c": 1, "b": 0, "f": 5, "h": 2, "d": 0, "p": 18, "v": 20.438, "a": "287A6E530600007B", "j": 0.0}]
2025-03-21 18:50:40 [N] Available devices received: [{"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 2, "d": 0, "p": 18, "v": 20.75, "a": "285EE23E060000F6", "j": 0.0}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 2, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 5, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 6, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 19, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 4, "x": 1}]
2025-03-21 18:50:59 [N] Installed devices received: [{"i": 0, "t": 1, "c": 1, "b": 0, "f": 5, "h": 2, "d": 0, "p": 18, "v": 20.438, "a": "287A6E530600007B", "j": 0.0}]
2025-03-21 18:50:59 [N] Available devices received: [{"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 2, "d": 0, "p": 18, "v": 20.75, "a": "285EE23E060000F6", "j": 0.0}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 2, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 5, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 6, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 19, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 4, "x": 1}]
2025-03-21 18:51:00 [N] Device updated to: {"i":1,"t":1,"c":1,"b":1,"f":9,"h":2,"d":0,"p":18,"a":"0000000000000000","j": 0.000}
2025-03-21 18:51:04 [N] Installed devices received: [{"i": 0, "t": 1, "c": 1, "b": 0, "f": 5, "h": 2, "d": 0, "p": 18, "v": 20.438, "a": "287A6E530600007B", "j": 0.0}, {"i": 1, "t": 1, "c": 1, "b": 1, "f": 9, "h": 2, "d": 0, "p": 18, "v": null, "a": "0000000000000000", "j": 0.0}]
2025-03-21 18:51:04 [N] Available devices received: [{"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 2, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 5, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 6, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 19, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 4, "x": 1}]
2025-03-21 18:52:34 [N] New JSON received from controller.
2025-03-21 18:54:16 [N] Resetting controller to factory defaults.
2025-03-21 18:54:30 [N] Resetting controller to factory defaults.
2025-03-21 18:54:35 [N] New JSON received from controller.
2025-03-21 18:54:53 [N] Installed devices received: []
2025-03-21 18:54:53 [N] Available devices received: [{"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 2, "d": 0, "p": 18, "v": 20.5, "a": "287A6E530600007B", "j": 0.0}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 2, "d": 0, "p": 18, "v": 20.75, "a": "285EE23E060000F6", "j": 0.0}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 2, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 5, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 6, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 19, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 4, "x": 1}]
2025-03-21 18:55:11 [N] Device updated to: {"i":0,"t":1,"c":1,"b":0,"f":5,"h":2,"d":0,"p":18,"a":"287A6E530600007B","j": 0.000}
2025-03-21 18:55:11 [N] Installed devices received: [{"i": 0, "t": 1, "c": 1, "b": 0, "f": 5, "h": 2, "d": 0, "p": 18, "v": 20.375, "a": "287A6E530600007B", "j": 0.0}]
2025-03-21 18:55:11 [N] Available devices received: [{"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 2, "d": 0, "p": 18, "v": 20.813, "a": "285EE23E060000F6", "j": 0.0}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 2, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 5, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 6, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 19, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 4, "x": 1}]
[... cut ...] p": 18, "v": null, "a": "0000000000000000", "j": 0.0}]
2025-03-21 18:55:29 [N] Available devices received: [{"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 2, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 5, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 6, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 19, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 4, "x": 1}]
pi@ultrasaha:~ $

lbussy, 2025-03-26:
Sorry, meant to reply and I got sidetracked.

This is a strange one; I'd say check all connections, see if you have another set of probes, and try those.  Also, try that Uno Test up above; it removes BrewPi from the equation.

I had this happen once and it was my pull-up resistor.  The leads were long-ish so I went to a stronger pull-up (2k2 as I remember) and it worked.

ultrasaha, 2025-03-30:
Dear Lee,

First of all, I would like to sincerely thank you for the support you provided. Although I tried to get my installation working with BrewPi Remix, I ultimately decided to switch to a more modern Raspberry Pi and set up Fermentrack instead. It’s now up and running smoothly.

I truly appreciated your assistance throughout the process—it made me feel supported every step of the way.

Thanks again for your help and dedication!

Best regards,

Ultrasaha.
Sr.J.

closed 2025-03-30 by ultrasaha

lbussy, 2025-03-30:
I am glad you got something running that works for you.  @thorrak and I work pretty closely together, and it's always been my intent to pressure him to finish version two so I can begin pushing people that way.

BrewPi, then Remox, is a well worn and rock solid path, but maintaining the python part of the software is challenging in this ecosystem.

Underneath the covers, Furman track is still using most of the parts of the original, so your migration should be smooth.